egzumer/uv-k5-firmware-custom #612: Airband channels

richard9999999999, 2024-09-09:
It is maybe rather a feature request than a bug. Probably you are aware that in aircraft communication, the published frequency number (number to dial on aircraft radio) is (can) be different from target frequency. More info e.g. here:
https://833radio.com/news/show/7
so e.g.:
- if you set the number 118.025 on the radio, radio will transmit/receive on 118.0250MHz with settings for 25kHz channel spacing
- if you set the number 118.030 on the radio, radio will transmit/receive on 118.0250MHz with settings for 8.33Khz channel spacing
- if you set the number 118.035 on the radio, radio will transmit/receive on 118.0333MHz with settings for 8.33Khz channel spacing
- if you set the number 118.040 on the radio, radio will transmit/receive on 118.0417MHz with settings for 8.33Khz channel spacing
- if you set the number 118.050 on the radio, radio will transmit/receive on 118.0500MHz with settings for 25Khz channel spacing
- if you set the number 118.055 on the radio, radio will transmit/receive on 118.0500MHz with settings for 8.33Khz channel spacing

I think it can be very convenient to implement this airband frequency dial mode in the K-5 firmware. You usually have some published "frequency" for airfield or aircraft service which in fact is not a frequency but a number to set on the airband radio..

prokrypt, 2024-09-09:
what is the current behavior? 

I would've thought this bit of code would be doing the right thing: 
[image]

richard9999999999, 2024-09-10:
uint32_t FREQUENCY_RoundToStep(uint32_t freq, uint16_t step)

{

                if(step == 833) {

        uint32_t base = freq/2500*2500;

        int chno = (freq - base) / 700;    // convert entered aviation 8.33Khz channel number scheme to actual frequency. 

        return base + (chno * 833) + (chno == 3);

                }

 

                if(step == 1)

                               return freq;

                if(step >= 1000) 

                               step = step/2;

                return (freq + (step + 1) / 2) / step * step;

}

 

 

I will speak about 8.33 spacing mode only now: that code  finds the nearest airband chanel frequency to the fiven frequency. 

In Airband,  e.g. 118.025 and 118.030 means the same frequency 118.025MHz (first in 25 khz spacing, second in 8.33kHz spacing)

Code above returns 

118.025 for 118.025

But

118.03333 for 118.030

In 8.33 spaced airband numbers set simply doesn’t corelate simply matematically with frequency. In the current firmware you need to know on which real frequency the given airport is however controlers, pilots and published documents do not use/state the frequency but the number which needs to be set to get desired frequency. I think i tis a good idea to implement airband mode to just need to set published numbers on radio..

 

Screenshot_20240909-094258-556.png (view on web) <https://github.com/user-attachments/assets/a32891c0-f406-4970-bb15-79778560af85>

richard9999999999, 2024-09-10:
..If somebody asks why 8.33 air band is so "unlogical", it is because historically 6 digit frequencies were reported/published/set on radios. After transfer to 8.33kHz channel spacing they wanted to save this plus also to have the possibility to distinguish between 25KHz and 8.33kHz spaced channels bu just setting the different number (despite the frequency is the same). Radio knows which spacing mode it is so it can switch the mode automatically. And yes, some service still uses 25kHz spacing in Europe: those are services which needs repeaters to have big coverage.. Those repeaters works on a bit shifted frequency from the base (each repeater on a bit different one) otherwise they would interfere badly each other in the range areas. Thus wider band (25KHz spacing) is needed here..

richard9999999999, 2024-09-10:
BTW I have communicated also directly with QuanSheng support on this topic but I was not successful. And their firmware has the 8.33 airband solved much worse: they simply added 8.33 to the base frequency 118MHZ :-)  So the errors were accounted badly with increasing frequency and even correct frequencies could not be set. What I was wondering was I never found any complaint on that topic on Net from any user. I simply put the radio on shelf for some year but now I tried to look if Quansheng have made some improvement. I didn't find the firmware on their site anymore but instead I found your work. Great!!!

g4eml, 2025-02-06:
This feature has been implemented for a long time. The code snippet you quote does exactly that. 

Your example of entering 118.025 and 118.030 works as it should. Both result in the radio being set to 118.025. 

Make sure that you have set the radio to 8.33Khz steps. The Airband channel mode only works correctly with that setting. 

Colin G4EML

richard9999999999, 2025-02-06:
Hi Colin, I confirm the code works for converting airband channel to frequency (I've made a mistake while testing and get self confused somehow...) Still I would vote for an "Airband mode" possibility where displayed number means the "Airband channel" not the real frequency. Something like a modus of work of real airband radio.