egzumer/uv-k5-firmware-custom #616: After getting back to ezumer this happans!?

rankokrneta, 2024-09-28:
Hi,

I hope this message finds you well.

I was eager to test some other firmware, and after trying Fagci (uvk5-spectrum-reborn-fagci-20240225_2215.bin ), I now want to revert to the latest EGZumer firmware. Unfortunately, I'm having trouble doing so. Despite my efforts, I’m stuck at the same point, as shown in this video: [YouTube Link](https://www.youtube.com/shorts/Y-zIaNOouAc).

I’ve tried various methods, including web-flash and using K5Tool-Win, but nothing seems to work. 

Is there a way to resolve this issue?

Thank you for your help!

Best regards,  
Ranko

prokrypt, 2024-09-29:
You need to restore your eeprom. Hopefully you made a backup or a download with chirp?

deadfred-, 2025-01-17:
Also getting the same issue with v22.  Tried restoring to stock image and restoring EEPROM and no luck...

kamkom21, 2025-04-12:
For future. If you have similiar issue only fix i found is:

1. flash ijv 3.6 firmware
2. turn off radio
3. turn radio on with holding s1 button
4. in settings search for reset select all
5. after radio resets and reboots you can install egzumer and restore your eeprom.