egzumer/uv-k5-firmware-custom #620: SCREEN FREEZE AND NEED TO TURN OFF AND ON TO RESET.

nicholasindoril, 2024-10-08:
HELLO MY FRIEND. THANK YOU FOR THIS FIRMWARE AND ALL YOUR WORK SO FAR. I HAVE NOTICED WHILE SCANNING AND FINDS A FREQUENCY AND STANDS THERE. AFTER USING THE PRESS BUTTON TO ENTER THE [LMA-s/LMA/PGA/IF] OPTION MENU AND USE THE B AND C BUTTONS TO STEP UP OR DOWN, WHEN REACHING THE UPPER OR DOWN LIMIT IT FREEZES THE SCREEN THERE AND NO BUTTON UNFREEZES IT. I HAVE TO TURN THE RADIO OFF AND ON AGAIN. 
MY VERSION IS "EGZUMER 7607f0a" AS SHOWN ON STARTUP.

SCREENSHOT:
[image]

prokrypt, 2024-10-09:
It's a pretty old bug. Temporary "fix" would be to remove antenna to unfreeze.

nicholasindoril, 2024-10-09:
> It's a pretty old bug. Temporary "fix" would be to remove antenna to unfreeze.

THANKS A LOT. IT WORKS!!