egzumer/uv-k5-firmware-custom #621: External PTT trouble

F1LYS, 2024-10-08:
When de-squelched (on value 0) cannot activate TX with external mike thru jack plugs. Works fine if squelch is activated (on value 1 or more) .

BruMcD, 2024-10-27:
A known hardware issue.   Basically along with the received audio many versions of this set send about 4V DC to the external speaker.  (Due to a sloppy/ill thought through design)    This overheats small earphones and interferes with External PTT operation (which relies on a DC level on the Speaker line which now cannot be achieved) too

The fix is a blocking capacitor in your speaker Mic (if you can be bothered)

F1LYS, 2024-10-30:
Hello, 
Thank you for your advice. I have inserted a blocking capacitor after checking the dc voltage presence when sql is 0; strange behavior!. Now it works.
Congratulation for the quality of your work on this project and many thanks for the time that all the team have spend on that matter.

BruMcD, 2024-10-30:
If you're curious the reason for the 3.5 - 4v is they decided to use a bridged Audio Output amp.  (Two power amps (fed out of phase) are connected to the two ends of the internal speaker which floats between them) 
This removes the need for a large (or Low frequency limiting) output blocking capacitor and in low voltage supply use may also increase audio power a bit.    At one stage they were pushing "improved Audio" over this mid-production cost saving . . .       

This is all well until  you spend a 1minute longer than the designer did on the Side-socket audio output.  The Amp is switched off when squelched,  but in RX (e.g. on a repeater)  both ends of the external speaker 'rest' at 3.5 - 4V (half supply)
The 'Kenwood' plug only has one pin for audio out and must use the chassis as reference.   They can't connect or mix the two speaker wires so just used one of them.  The full audio waveform is there on receive but so is the 4V baseline.   
The difference between the two ground sleeves on the 2 plugs signals PTT to the radio - but this speaker arrangement feeds low impedance 4V to them via the speaker coil

So Ext PTT operation during RX with typical speaker mics is interfered with by the DC voltage.   
If you press PTT before the receiver is open it works.     Early manuals included this as a feature :-)

gretel, 2024-11-07:
glad i've found this thread. tried capacitor on `MIC+` and `GND` first but it had to go on `PTT` and `GND` to fix this issue. regards