egzumer/uv-k5-firmware-custom #626: Radtel rt 4d

4ndy1, 2024-11-09:
Will there be anyone working on improving airband on this radio at all it’s quite impressive radio vhf uhf and airband and dmr all in one…RADTEL RT 4d

Alex-Roe, 2024-11-11:
Check the radtel website, there is already a firmware fix for AM

4ndy1, 2024-11-11:
Yep your right but not very good it drifts in and out of clear audio some work still needing to be done

4ndy1, 2024-11-11:
I hope they can do with this what they did to the quansheng then this will be an excellent radio covering airband marine band and dmr

tick-hack, 2024-11-28:
hi, how you update firmware? Because i stuck at 2\246 percent (CH340+ windows 10)

wuebas, 2024-12-06:
It needs a custom firmware that allows it to scan faster, improvements such as SSB in AM mode and in the future the possibility of adding a SI4732 card to be able to listen to HF as well.

tinuva, 2025-04-03:
This is a DMR radio, don't expect it to run this firmware. Rather go ask over at opengd77 project which is for dmr radios, in my opinion that is the correct path to ask.

AntonioAC230, 2025-04-03:
Since April 2025, an update has been available that introduced the ability to receive SSB and AM, as well as a promiscuous mode in DMR. The radio operates in the 18–1000 MHz range (albeit with a few minor gaps). I personally often switch the antenna in my car and listen to channel 19 (27.180 MHz AM in Poland). The output power at this frequency is a drozen of so milliwatt range, and the signal contains many harmonics – similar to Quansheng outside the band 2m/70cm. It is a cool, cheap, and universal radio. The firmware has some shortcomings, but the direction of its development is interesting. Another point is that only after this update did the radio become usable—something that could not be said before due to the limited 16-position RX list.

vk3xem, 2025-06-01:
Can the RT4D be made to TX up to 480 MHz?