egzumer/uv-k5-firmware-custom #636: Lock Rx frequencies

jon-daemon, 2024-12-24:
I just got my UV-K5 (99) and the first thing I did was backup and flash egzumer.
I was able to unlock/lock frequencies using CHIRP and the hidden menu (61).

I was wondering if it's possible to lock receiving on the same frequencies that transmitting is locked. I want to get the radio with me outdoors and I want it to be locked on ham frequencies when traveling for legal reasons. On a police check I might be in trouble even if my radio can listen to frequencies out of the ham bands.

Would that be possible?

mentanah, 2024-12-25:
Just Set in chirp to "memory Channels only" the free frequency selection is not available...

jon-daemon, 2024-12-25:
It would be great if it could be done through the hidden menu so we could change that feature on the go without a computer.