egzumer/uv-k5-firmware-custom #638: error compiling

EXCISIONMODZ, 2024-12-29:
/usr/lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: firmware section `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 700 bytes
collect2: error: ld returned 1 exit status
make: *** [makefile:441: firmware] Error 1

JuantAldea, 2024-12-29:
Did you activate any additional features?

EXCISIONMODZ, 2024-12-29:
https://prnt.sc/uE1unoJqh-35
heres the make file beginning

ze-nite, 2025-04-30:
its happening to me too

`arm-none-eabi-gcc -z noexecstack -mcpu=cortex-m0 -nostartfiles -Wl,-T,firmware.ld -Wl,--gc-sections --specs=nano.specs start.o init.o external/printf/printf.o driver/adc.o driver/aes.o driver/backlight.o driver/bk1080.o driver/bk4819.o driver/crc.o driver/eeprom.o driver/gpio.o driver/i2c.o driver/keyboard.o driver/spi.o driver/st7565.o driver/system.o driver/systick.o driver/uart.o app/action.o app/aircopy.o app/app.o app/chFrScanner.o app/common.o app/dtmf.o app/flashlight.o app/fm.o app/generic.o app/main.o app/menu.o app/spectrum.o app/scanner.o app/uart.o am_fix.o audio.o bitmaps.o board.o dcs.o font.o frequencies.o functions.o helper/battery.o helper/boot.o misc.o radio.o scheduler.o settings.o ui/aircopy.o ui/battery.o ui/fmradio.o ui/helper.o ui/inputbox.o ui/main.o ui/menu.o ui/scanner.o ui/status.o ui/ui.o ui/welcome.o version.o main.o -o firmware
/usr/lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: firmware section `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 604 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:441: firmware] Error 1`

JuantAldea, 2025-08-04:
The memory is almost fully-used and whenever you enable a feature, you have to disable another. Otherwise the FW does not fit in memory.