egzumer/uv-k5-firmware-custom #644: Running v0.21 - A request for the USB device descriptor failed.

rschiefer, 2025-01-24:
I'm trying to upgrade to 0.22 from 0.21 but my laptop won't find the com port.

The cable works fine on another radio and i'm pressed it hard when connecting.  Tried from flash mode (holding PTT when powering on) and normal neither work.

[image]

prokrypt, 2025-01-24:
This is a hardware issue. Try another cable, another port, or insert a USB hub in between.

rschiefer, 2025-01-24:
Very thankful for your reply @prokrypt!

I've tried 2 cables and both worked with a different radio. Tried multiple ports on multiple laptops as well as going through a usb hub.  Same result each time. 

All indications are it has something to do with the firmware on the radio (Egzumer 0.21).  

Any other ideas?

prokrypt, 2025-01-24:
What happens if you plug just the cable into the computer? The serial port
should still show up even if the radio isn't connected...

rschiefer, 2025-01-24:
Nothing happens when I plug only the cable into the computer.  Once I turn on the radio (either in normal mode or flash mode) I get the above error.

Both cables work correctly on my Anyton 878 and show a COM port once I turn the radio on.

prokrypt, 2025-01-24:
The cable for the 878 is not a serial adapter. You'll need to get a proper serial adapter cable for the Quansheng (and other Baofeng/cheap radios)

rschiefer, 2025-01-24:
You are right.  I cracked open the case on both cables and neither have the ftdi chip they were just made to look like they did.  Ordering a legit serial ftdi cable.

This is also a good resource for others having similar issues - https://www.miklor.com/COM/UV_GettingStarted.php

Thanks again @prokrypt !