egzumer/uv-k5-firmware-custom #647: Concerning Apache 2.0 License Compliance of UV-K5-RX-JP Firmware

BB-KING777, 2025-04-04:
Hello to the community and maintainers,

I would like to bring to the community's attention a potential license compliance issue with a derivative project based on this firmware.

The project in question is "UV-K5-RX-JP" (https://weboo.dev/posts/uv-k5-firmware-rx-jp/), which is a Japanese localized version of this firmware with additional modifications. This derivative work states that it's based on egzumer's firmware.

However, this derivative project imposes restrictions that appear to conflict with the Apache 2.0 license:

"※ Redistribution or transfer to third parties, whether for a fee or free of charge, without the permission of the copyright holder is prohibited. It cannot be sold as a product or part of a product on auction or flea market sites. If you intend to use it for commercial purposes, you need to enter into a separate commercial license."

These restrictions conflict with the Apache 2.0 license terms in several ways:
1. The Apache 2.0 license permits redistribution (both commercial and non-commercial)
2. It allows commercial use without requiring a separate license
3. It requires that modified versions maintain the same license terms

Additionally, it appears that the source code for this derivative work might not be publicly available, particularly for the latest versions.

I'm bringing this to the community's attention to:
1. Seek clarification on whether this is indeed a license violation
2. Determine what the community's stance is on such derivative works
3. Discuss potential approaches to address these kinds of licensing issues

I believe maintaining proper license compliance is important for the health and sustainability of open-source projects.

Thank you all for considering this matter.

maxbrito500, 2025-05-02:
The Apache-2.0 license itself *does permit to build products which are closed source*. It is different, from example, of licenses like the GPL which are known for copyleft impact and will force the product using them to be free software too.

In that sense, the author of that japanese product is *not* selling you egzummer. He is selling egzumer + own changes. So you are basically paying a license for whatever code he placed on top of something that was Apache-2.0.

That business model is similar to Red Hat Linux. Where even with a copyleft license on Linux you are basically paying Red Hat access for Linux plus the code written by Red Hat (plus support). There isn't much that can be done on this case.

BB-KING777, 2025-05-05:
Thank you for acknowledging this perspective. I think we're on the same page now regarding the license compliance issue.

The next logical step would be to contact the author of the UV-K5-RX-JP project to clarify:
1. Whether they're aware of the Apache 2.0 license requirements for modified source code
2. If they can bring their project into compliance by either:
   - Releasing their modified source code under Apache 2.0
   - Clearly separating which components are Apache 2.0 licensed and which are proprietary

I'll try reaching out to them first with a friendly inquiry rather than immediately assuming non-compliance. Perhaps there's been a misunderstanding about license requirements or there might be technical reasons we're not aware of.

I appreciate your open-minded approach to this discussion. The goal isn't to create conflict but to ensure the open-source ecosystem functions properly with appropriate license compliance.

Thank you for caring about this issue.