egzumer/uv-k5-firmware-custom #651: Spectrum Scanner possible bug

ze-nite, 2025-05-13:
I´ll just leave my reddit post with the problem here, it has a video of it, its way easier to understand.

https://www.reddit.com/r/amateurradio/comments/1kl7lvo/uvk5_possible_bug
https://www.reddit.com/r/amateurradio/comments/1kl7lvo/uvk5_possible_bug
https://www.reddit.com/r/amateurradio/comments/1kl7lvo/uvk5_possible_bug

prokrypt, 2025-05-13:
It's more of a hardware bug, but sometimes those can be worked around via software. For this particular one it's probably easier to just add the offending frequency to the skip list or bump up the squelch...

ze-nite, 2025-05-13:
thank you, so is this like a known thing and not really a problem with my radio?

I also think it might be the type of signal because I moved to another town today and this didn’t happen to me before

prokrypt, 2025-05-13:
Yep, it's normal. I have encountered the same thing on a Yaesu FT1XD as well as FT3DR, usually related to the LCD screen updates but sometimes due to the internal oscillator PLL  (they have a software workaround but it doesn't always work).